facebookresearch/home-robot #464: KeyError in demo running

cazhang, 2024-01-14:
1. Tried to run the demo: 
python projects/habitat_ovmm/eval_baselines_agent.py --env_config projects/habitat_ovmm/configs/env/hssd_demo.yaml

2. Got KeyError: 
  File "/home/czhang/home-robot/src/home_robot/home_robot/perception/wrapper.py", line 126, in _process_obs
    obs.task_observations["start_recep_goal"] = self.vocabulary_name_to_id[
KeyError: 'cabinet'

3. Checked self.vocabulary_name_to_id: 
self.vocabulary_name_to_id: {'bowl': 1, 'chair': 2, 'table': 3}
4. Checked the target hssd_demo.yaml, it has: 
start_recep: "chair"
pick_object: "bowl"
goal_recep: "table"

Question: 'cabinet' was not matching any of the vocabulary, wondering what caused this unexpected error?

yvsriram, 2024-01-18:
Hi, thanks for the issue. Yes, you are right. Due to a few recent changes this hssd_demo.yaml config works only in real world. You can use hssd_eval.yaml instead.

To make it work you can remove the `start_recep`, `pick_object`, `goal_recep` keys from the config. These will be picked from episodes. I will push a fix right now to ensure that the passed-in values are only used in the real world.

cazhang, 2024-01-18:
thanks. closing.

cazhang, 2024-01-18:
thanks. closing.

cazhang, 2024-01-18:
thanks. closing now.

cazhang, 2024-01-18:
thanks. close.

closed 2024-01-18 by cpaxton