facebookresearch/home-robot #465: Pretrained RL agent loading error

cazhang, 2024-01-15:
## 🐛 Bug

## Steps to Reproduce
Hi, I tried to evaluate the RL agent with the main branch. 
Command: projects/habitat_ovmm/eval_baselines_agent.py --baseline_config_path projects/habitat_ovmm/configs/agent/rl_agent.yaml

The pretrained checkpoints located in: data/checkpoints/ovmm_baseline_home_robot_challenge_2023_v0.2

## Errors:
Line 177 of home_robot/agent/ovmm_agent/ppo_agent.py gave errors and after a check: 
nav_to_obj.pth and nav_to_rec.pth could NOT be loaded using torch.load()

Could you please have a check on this?

asfandasfo, 2024-01-18:
hey, have you solved this? I'm getting this error while loading pretrained models. i ma using latest main with `rl_agent`

`Can't get attribute 'OVMMNavToObjExploreRewardMeasurementConfig' on <module habitat.config.default_structured_configs' from '/home/ayaar/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/config/default_structured_configs.py'>`

cazhang, 2024-01-18:
The pretrained checkpoints from v0.2 seems not correct !
So I switched to use the release version code, as well as its checkpoints, for evaluation purpose.
https://github.com/facebookresearch/home-robot/tree/home-robot-ovmm-challenge-2023-v0.1.2

cpaxton, 2024-01-18:
https://huggingface.co/datasets/ovmm/baseline_checkpoints/tree/main they should be uploaded here now as well

closed 2024-02-20 by cazhang