facebookresearch/home-robot #469: success_distance for FindObj module

asfandasfo, 2024-01-18:
Hey, can you point me to the right `config` file to change the  `success_distance` for `nav_to_obj` part? I tried severals but no effect.

yvsriram, 2024-01-18:
Hi, you can change it here: https://github.com/facebookresearch/habitat-lab/blob/45aa489a84a853cb10b9e7c87383262831a6bd22/habitat-lab/habitat/config/habitat/task/ovmm/ovmm.yaml#L118

asfandasfo, 2024-01-18:
@yvsriram I try changing it here

> Hi, you can change it here: https://github.com/facebookresearch/habitat-lab/blob/45aa489a84a853cb10b9e7c87383262831a6bd22/habitat-lab/habitat/config/habitat/task/ovmm/ovmm.yaml#L118

I try changing it here but no effect.

yvsriram, 2024-01-18:
> I try changing it here but no effect.

Can you please provide details of what you are testing and what effect you expect?

For instance, I tried setting the distance to a large value:

```
python projects/habitat_ovmm/eval_baselines_agent.py --baseline_config_path projects/habitat_ovmm/configs/agent/rl_agent.yaml habitat.environment.max_episode_steps=1 habitat.task.measurements.nav_orient_to_pick_succ.success_angle_dist=100 habitat.task.measurements.nav_orient_to_pick_succ.min_object_coverage_iou=-1 habitat.task.measurements.nav_to_pick_succ.success_distance=100
```

and I get ovmm_find_object_phase_success = 1.0

Note: here I increased success_distance to a very large value for a sanity check and hence had to increase the orientation thresholds as well (as the object may not be in view at start of episode).

asfandasfo, 2024-01-18:
> Can you please provide details of what you are testing and what effect you expect?
>
> For instance, I tried setting the distance to a large value:
>
> ```
> python [...]

I was trying to complete the `nav_to_obj` skill and start the 'gaze' skill when the robot is within 2m distance to object. I know it will not be able to pick the object but I was trying some experiments.
I will check it again.

yvsriram, 2024-05-03:
Closing. Please reopen if you have more questions.

closed 2024-05-03 by yvsriram