facebookresearch/home-robot #471: New episode generator

cxzohain, 2024-01-22:
Hi, Is there a way to generate a new episode but with the specific `goal_object` and `receptacles`. E.g. `goal_object:cup, start_receptacle:counter, goal_receptacle: stool`

yvsriram, 2024-01-31:
Hey! If I understand you correctly, you are trying to filter episodes by the category of goal object, start receptacle and target receptacle (and not trying to generate new episodes).

You can first identify the episode ids using

```
import json, gzip
episodes = json.load(gzip.open('data/datasets/ovmm/val/episodes.json.gz'))['episodes']

filtered_ids = [episode['episode_id'] for episode in episodes if episode['object_category'] == 'cup' and episode['start_recep_category'] == 'counter']
```

and then evaluating by passing in these ids:

`python projects/habitat_ovmm/eval_baselines_agent.py <other options> habitat.dataset.episode_ids=['1','2']`

cxzohain, 2024-01-31:
I am trying to use some receptacles that are already present in the scenes but not used in the provided episodes data. E.g. in the scene `104348328_171513363` the `bed` is present but there is no episode which have `bed` as `start_receptacle` or `goal_receptacle` for this scene. How can I generate such episode?

yvsriram, 2024-02-05:
Got it. Please check out the config file I shared above: https://github.com/facebookresearch/habitat-lab/blob/ovmm_main_merge/habitat-lab/habitat/datasets/rearrange/configs/ovmm_val.yaml (or ovmm_train.yaml).

First, restrict the `scene_sets` to a single scene. Then, update receptacle_sets to include the receptacles of interest. For this, you can filter receptacle names by category using this mapping: https://huggingface.co/datasets/ai-habitat/OVMM_objects/blob/main/hssd-receptacles.csv. For example, for bed include all substrings like "0081c522ffe7910c701191c44b0f797a69a8548a"

yvsriram, 2024-03-05:
Close the issue. Feel free to let us know if you have more questions.

closed 2024-03-05 by yvsriram

yvsriram, 2025-07-28:
Thanks for the issue! Yes that branch got deleted. Please replace ovmm_main_merge with ovmm_v0.2.5 in the link