facebookresearch/home-robot #472: Error while training nav_to_rec skill

cxzohain, 2024-01-27:
Hi, I am trying to train `nav_to_rec` skill and getting this error. Can you check? Thanks
` return F.linear(input, self.weight, self.bias)
RuntimeError: mat1 and mat2 shapes cannot be multiplied (20x1 and 4x32)`

yvsriram, 2024-01-31:
This is not expected. I will check. Feel free to share the command you were using.

cxzohain, 2024-02-01:
> This is not expected. I will check. Feel free to share the command you were using.

`python -u -m habitat_baselines.run --config-name  /home/home-robot/src/third_party/habitat-lab/habitat-baselines/habitat_baselines/config/ovmm/rl_skill.yaml habitat_baselines.evaluate=False benchmark/ovmm=nav_to_obj`

yvsriram, 2024-02-05:
Thanks! Please use  `rl_discrete_skill.yaml` (instead of `rl_skill.yaml`) when training nav_to_obj and nav_to_rec policies. This changed after an upgrade - I will update the training instructions.

```
python -u -m habitat_baselines.run    --config-name=ovmm/rl_discrete_skill.yaml habitat_baselines.evaluate=False   benchmark/ovmm=nav_to_obj habitat_baselines.checkpoint_folder=data/new_checkpoints/ovmm/nav_to_obj
```

libaiwanovo, 2024-03-29:
When I run "python -u -m habitat_baselines.run --config-name=ovmm/rl_discrete_skill.yaml habitat_baselines.evaluate=False benchmark/ovmm=nav_to_obj habitat_baselines.checkpoint_folder=data/new_checkpoints/ovmm/nav_to_obj", it crashes immediately after reaching "agent number of parameters: 9472337" and displays "killed". My device is RTX 3090 with 24GB of memory.
Where should I make modifications to reduce the consumption?Looking forward to your answer.

yvsriram, 2024-03-29:
Hi @libaiwanovo , thanks for your interest! You can try reducing the number of environments per gpu. `habitat_baselines.num_environments=X`. We currently have this as 16. However, this might reduce your task performance unless you use more gpus increasing the effective number of environments. 

Let me know if you have any further questions.

libaiwanovo, 2024-03-30:
> Hi @libaiwanovo , thanks for your interest! You can try reducing the number of environments per gpu. `habitat_baselines.num_environments=X`. We currently have this as 16. However, this might reduce [...]

Thank you, I am trying.

yvsriram, 2024-05-03:
Closing, please reopen if you still have issues.

closed 2024-05-03 by yvsriram