facebookresearch/home-robot #473: Cannot install dependencies of home-robot

Afsanay, 2024-01-31:
## 🐛 Bug

[image]

# Error Description

I am not able to install the dependencies of the home robot while trying to setup the environment.
There were different errors first
ERROR 1 - no module named pypandoc
RESOLVED BY - pip install pypandoc

ERROR 2 - pypandoc has no attribute called convert
RESOLVED BY - pip install pypandoc==1.5

## Steps to Reproduce

1) Setup the new environment from the environment.yml file in the src/home_robot/environment.yml file
2) set the HOME_ROBOT_ROOT and CUDA_HOME environment variables
3) either:
  a) run "./install_deps.sh"
  or:
  b) go in the step by step instructions and then install pinocchio and finally run  "pip install -e src/home_robot"

## Expected behavior

The environment should be able to run the command and set up the "home_robot" module in the environment.

yvsriram, 2024-01-31:
I tried a fresh installation and `install_deps.sh` works for me without any errors. I followed the commands you listed. The only difference is I used mamba as per the instructions. 

Also, I did not need to install `pandoc` in my environment. Do you happen to remember which dependency required `pandoc` in your case?

yvsriram, 2024-01-31:
Ubuntu 20.04.6 LTS
mamba 1.4.4
conda 23.7.2

closed 2024-02-09 by Afsanay