facebookresearch/home-robot #475: Error when submodule loaded by git

Moresweet, 2024-03-01:
## 🐛 Bug

Hello, I get "Page not found" when clicking src/third_party/spot-sim2real @ 1cecb34. And the bash file install_deps.sh will report error until row 79. Although I modify the .gitmoudles from Jdvakil/spot-sim2real to facebookresearch/spot-sim2real, it still not work. the branch called no-habitat havn't been found.
[image]

## Steps to Reproduce

Steps to reproduce the behavior:

command:
`./install_deps.sh`

## Expected behavior

fatal: 远程错误：upload-pack: not our ref 1cecb34f5a3e52119baec35bbc9d48bf48e06431
获取了子模组路径 'src/third_party/spot-sim2real'，但是它没有包含 1cecb34f5a3e52119baec35bbc9d48bf48e06431。直接获取该提交失败。

## Additional context

I think it is caused by the project sim2real has moved to facebookresearch/sim2real while the home-robot's submodule is old.

yvsriram, 2024-03-02:
@Jdvakil It seems like the referenced commit (`1cecb34`) is private. Is there a public commit we could use here?

Also, @Moresweet this submodule should not be necessary if you are not working with real Spot.

Jdvakil, 2024-03-05:
Checking it out

Jdvakil, 2024-05-16:
Should be fixed with #495

closed 2024-05-16 by Jdvakil