facebookresearch/home-robot #477: Issue with running Habitat OVMM

ramkumarkoppu, 2024-03-02:
## 🐛 Bug

Issue with running Habitat OVMM demo

## Steps to Reproduce

Steps to reproduce the behavior:

when I run 
`python projects/habitat_ovmm/eval_baselines_agent.py --env_config projects/habitat_ovmm/configs/env/hssd_demo.yaml` provided in the https://github.com/facebookresearch/home-robot/blob/main/projects/habitat_ovmm/README.md, I 

got following error:
`pybullet build time: Nov 28 2023 23:52:03
Traceback (most recent call last):
  File "/home/ram/repos/home-robot/projects/habitat_ovmm/eval_baselines_agent.py", line 10, in <module>
    from evaluator import OVMMEvaluator
  File "/home/ram/repos/home-robot/projects/habitat_ovmm/evaluator.py", line 18, in <module>
    from utils.env_utils import create_ovmm_env_fn
  File "/home/ram/repos/home-robot/projects/habitat_ovmm/utils/env_utils.py", line 13, in <module>
    from home_robot_sim.env.habitat_ovmm_env.habitat_ovmm_env import (
  File "/home/ram/repos/home-robot/src/home_robot_sim/home_robot_sim/env/habitat_ovmm_env/habitat_ovmm_env.py", line 35, in <module>
    from home_robot_sim.env.habitat_objectnav_env.visualizer import Visualizer
  File "/home/ram/repos/home-robot/src/home_robot_sim/home_robot_sim/env/habitat_objectnav_env/visualizer.py", line 19, in <module>
    from home_robot.mapping.instance.instance_map import InstanceMemory
  File "/home/ram/repos/home-robot/src/home_robot/home_robot/mapping/__init__.py", line 5, in <module>
    from .voxel import SparseVoxelMap, SparseVoxelMapNavigationSpace
  File "/home/ram/repos/home-robot/src/home_robot/home_robot/mapping/voxel/__init__.py", line 7, in <module>
    from .voxel_map import SparseVoxelMapNavigationSpace
  File "/home/ram/repos/home-robot/src/home_robot/home_robot/mapping/voxel/voxel_map.py", line 17, in <module>
    from home_robot.utils.geometry import angle_difference, interpolate_angles
  File "/home/ram/repos/home-robot/src/home_robot/home_robot/utils/geometry/__init__.py", line 7, in <module>
    from ._base import *
  File "/home/ram/repos/home-robot/src/home_robot/home_robot/utils/geometry/_base.py", line 49, in <module>
    def xyt2sophus(xyt: np.ndarray) -> sp.SE3:
AttributeError: module 'sophus' has no attribute 'SE3'`

my python env, got this:
`conda list | grep -i sophus
sophus                    23.0.1                   pypi_0    pypi
sophuspy                  1.2.0                    pypi_0    pypi`

## Expected behavior

demo should be running without any errors

## Additional context

yvsriram, 2024-03-02:
We previously had 0.0.8 version installed. It seems like the library got updated recently. Can you please try updating your sophuspy version to 0.0.8: `pip install sophuspy==0.0.8`? Please let us know if you run into any other issues.

yvsriram, 2024-03-05:
The issue should be resolved. Feel free to reopen if the issue persists.

Rongtao-Xu, 2024-04-16:
My sophuspy is 0.0.8 ,  But still get the error:
home_robot/utils/geometry/_base.py", line 49, in <module>
    def xyt2sophus(xyt: np.ndarray) -> sp.SE3:
AttributeError: module 'sophus' has no attribute 'SE3'
Please tell me if you can solve it?

aliciaji1993, 2024-07-11:
> My sophuspy is 0.0.8 , But still get the error: home_robot/utils/geometry/_base.py", line 49, in def xyt2sophus(xyt: np.ndarray) -> sp.SE3: AttributeError: module 'sophus' has no attribute 'SE3' Please tell me if you can solve it?

instead of "import sophus as sp", use "import sophuspy as sp"