facebookresearch/home-robot #479: Moving agent to particular gps and compass

yrraadi-io, 2024-03-04:
Hi, I was wondering if this feature already exists. I'm currently collecting the `observations.gps` and `observations.compass `values in the `evaluator.py` file. I want to be able to spawn my agent at a particular gps and compass I collected during one of the episodes.

How can I do that? Does the feature exist?

yvsriram, 2024-03-04:
No, the feature doesn't exist currently. 

But happy to help with the changes you will need to make.

Firstly, the gps and compass values recorded here are relative to the episode's start pose. I think it may be much easier to 
 directly record the absolute pose with new sensors (look at how the [current relative pose sensors](https://github.com/facebookresearch/habitat-lab/blob/ovmm_main_merge/habitat-lab/habitat/tasks/rearrange/sub_tasks/nav_to_obj_sensors.py#L241..L276) are implemented) and then collect them in evaluator.py. You can set the absolute pose to the desired values in the episode reset function (similar to [this](https://github.com/facebookresearch/habitat-lab/blob/45aa489a84a853cb10b9e7c87383262831a6bd22/habitat-lab/habitat/tasks/rearrange/sub_tasks/nav_to_obj_task.py#L243) where we read the pose from episodes) and optionally create a config param for passing them in.

yrraadi-io, 2024-03-04:
Thank you for sharing this. Basically our goal is to directly have the agent spawn facing a candidate object and we want to experiment on a few things with respect to detection.
What's the best way you recommend we do that? Do we read the absolute pos of candidate objects from the val episode files or do we run the simulation till we find a candidate object, log its absolute pos and then save it?

yvsriram, 2024-03-26:
Closing this. Let us know if you have any more questions.

closed 2024-03-26 by yvsriram