facebookresearch/home-robot #481: Difference between first two components of snapshot image

yrraadi-io, 2024-03-26:
I'm confused about the difference between the first two components in the `snaspshot_<time_step>` image being saved after each episode's run.

[image]
Above are the two side-by-side images i'm talking about.

They both seem to be depicting object detection confidence score + mask segmentation.
Appreciate any help understanding this diff, thanks!

yvsriram, 2024-03-26:
The right one just overlays the mask - for each predicted object Detic also provides us with masks. If you do not like the box around the object in the left frame, you can try setting `use_detic_viz=False` https://github.com/facebookresearch/home-robot/blob/5eca3968af0451058795c8e160600aa50ccf0a67/projects/habitat_ovmm/configs/env/hssd_demo.yaml#L22

yrraadi-io, 2024-03-26:
Hi, that makes sense. two questions here:
1. The left frame seems to have a beige color mask too over the duck.
2. I want to replace the left frame with the ground truth masks, how do you suggest I do this?

yvsriram, 2024-03-28:
Hi,

1. I think the beige color mask comes from the Detic visualisation too. Setting `use_detic_viz = True` should remove it too.
2. If `env_config.ground_truth_semantics = False` (which is the case for you), GT is not passed along - fields like `object_segmentation` have detector outputs. To visualize both, firstly you can add separate GT fields for these observations [here](https://github.com/facebookresearch/home-robot/blob/5eca3968af0451058795c8e160600aa50ccf0a67/src/home_robot_sim/home_robot_sim/env/habitat_ovmm_env/habitat_ovmm_env.py#L193). Let's call them `gt_object_segmentation`. Then include it in the info dict [here](https://github.com/facebookresearch/home-robot/blob/5eca3968af0451058795c8e160600aa50ccf0a67/src/home_robot/home_robot/agent/objectnav_agent/objectnav_agent.py#L467). This dict will get passed to visualize function where you can overlay this on top of the rgb person frame  [here](https://github.com/facebookresearch/home-robot/blob/5eca3968af0451058795c8e160600aa50ccf0a67/src/home_robot_sim/home_robot_sim/env/habitat_objectnav_env/visualizer.py#L445).

yrraadi-io, 2024-03-28:
thank you. that worked. had to tweak a few more things, but ground truth is displaying now:
[image]

closed 2024-03-28 by yrraadi-io