facebookresearch/home-robot #485: Question Regarding GOAT Code Branches

www-Ye, 2024-04-01:
Hello, I'm very interested in your work on GOAT.
I noticed that there are several branches of code related to "goat" in the repository. Which one is the complete implementation of the article "GOAT: GO to Any Thing," and are there any differences between them?

zhengsipeng, 2024-04-02:
I have the same question, do you release the ultimate version of GOAT?

Welly0902, 2024-04-26:
Same question here.

yvsriram, 2024-05-03:
Hey,  `ray_tracing` branch should have the latest code for the experiments we ran on the Spot Robot. This should be the entry point: https://github.com/facebookresearch/home-robot/blob/ray_tracing/projects/spot/goat.py.

If you are interested in sim benchmarking, this other work may be relevant: https://github.com/Ram81/goat-bench