facebookresearch/home-robot #486: How to use my own data?

libaiwanovo, 2024-04-02:
Thank you for the excellent work. I'm currently trying to adapt it for my own data. I have a Songling robot. How can I replace it with data from the Songling robot and my own environment?

yvsriram, 2024-04-11:
Hi, that sounds really exciting. You can use our perception stack and agent API as long as you implement your low-level stack to use the action and observations interfaces: https://github.com/facebookresearch/home-robot/blob/main/src/home_robot/home_robot/core/interfaces.py. You will have to build your own "AgileX" environment for this. You can look up our Stretch's environment as a reference: https://github.com/facebookresearch/home-robot/blob/main/src/home_robot_hw/home_robot_hw/env/stretch_pick_and_place_env.py

libaiwanovo, 2024-04-11:
Thank you, I'm trying this out, and once again, thanks for your work!

closed 2024-04-11 by libaiwanovo