facebookresearch/home-robot #490: Trouble Downloading Dataset

eyucherin, 2024-04-29:
## 🐛 Bug

I am trying to participate in the home-robot challenge and have followed the instructions [here](https://github.com/facebookresearch/home-robot/blob/home-robot-ovmm-challenge-2023-v0.1.2/docs/challenge.md). Everything works up until step 5. Step 6 tells me to download the required data. When I sign into hugging face there is no place to accept the license which won't let me proceed to next steps.

[image]

When I try to download the scenes, I get the following. 

```
(habitat) che@che-MS-7D99:~/Desktop/home-robot/projects/habitat_ovmm$  ./install.sh
Download the scenes...
error: pathspec 'data/hssd-hab' did not match any file(s) known to git
Download the objects...
error: pathspec 'data/objects' did not match any file(s) known to git
Download the episodes...
error: pathspec 'data/datasets/ovmm' did not match any file(s) known to git
Safety; make sure you have downloaded everything.
If these are empty, you may have too old a git version.
./install.sh: line 26: cd: /data/hssd-hab: No such file or directory
Could not pull

Errors logged to '/home/che/.git/lfs/logs/20240429T223347.923422473.log'.
Use `git lfs logs last` to view the log.
/home/che
./install.sh: line 30: cd: /data/objects: No such file or directory
Could not pull

Errors logged to '/home/che/.git/lfs/logs/20240429T223347.969110534.log'.
Use `git lfs logs last` to view the log.
/home/che
./install.sh: line 34: cd: /data/datasets/ovmm: No such file or directory
Could not pull

Errors logged to '/home/che/.git/lfs/logs/20240429T223348.014118667.log'.
Use `git lfs logs last` to view the log.
```

## Steps to Reproduce

Steps to reproduce the behavior:
Follow steps 1~5 in 
[here](https://github.com/facebookresearch/home-robot/blob/home-robot-ovmm-challenge-2023-v0.1.2/docs/challenge.md) and try to download scenes without accepting to license.

## Expected behavior
I should be able to see a place to accept the license for using HSSD scenes in hugging face to download scenes. 

## Additional context
I was able to get a test version of habitat sim to work.

yvsriram, 2024-04-29:
Thanks you for participating! Happy to help you set up. Can you try downloading outside of the script. `cd data` and  `git clone https://huggingface.co/datasets/hssd/hssd-hab`? Does that work? Entering your huggingface credentials may be necessary.

Also please use [this branch](https://github.com/facebookresearch/home-robot/tree/docker-2024) and [this leaderboard](https://eval.ai/web/challenges/challenge-page/2278/overview) if you want to participate in the CVPR challenge.

eyucherin, 2024-04-30:
Hi! @yvsriram  Thank you so much for the reply! I just ran `git clone https://huggingface.co/datasets/hssd/hssd-hab` on `cd data`  and I was able to bypass `$HOME_ROBOT_ROOT/projects/habitat_ovmm/install.sh`  part of the dataset setup for OVMM.
However, I still am not able to  run the demo setup. 

ALSO:
When setting up for home_robot_sim, there seems to be an error when downloading pytorch3d. 
```
(home-robot) che@che-MS-7D99:~/Desktop/home-robot$ python -m pip install "git+https://github.com/facebookresearch/pytorch3d.git"
Collecting git+https://github.com/facebookresearch/pytorch3d.git
  Cloning https://github.com/facebookresearch/pytorch3d.git to /tmp/pip-req-build-a5a44a_w
  Running command git clone --filter=blob:none --quiet https://github.com/facebookresearch/pytorch3d.git /tmp/pip-req-build-a5a44a_w
  Resolved https://github.com/facebookresearch/pytorch3d.git to commit 47d5dc88247035b35ca3cfce159565f92d8fbb75
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [8 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-req-build-a5a44a_w/setup.py", line 15, in <module>
          import torch
        File "/home/che/.conda/envs/home-robot/lib/python3.9/site-packages/torch/__init__.py", line 218, in <module>
          from torch._C import *  # noqa: F403
      ImportError: /home/che/.conda/envs/home-robot/lib/python3.9/site-packages/torch/lib/libtorch_cpu.so: undefined symbol: iJIT_NotifyEvent
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

```

Any advice on this would be greatly appreciated!

yvsriram, 2024-04-30:
Can you try installing it after running `mamba install mkl=2024.0`.

eyucherin, 2024-05-01:
I am still getting erros after running `mamba install mkl=2024.0` 

for installing `pytorch3d` for the **home-robot simulator,** I still get the following error: 
```     
running build_ext
      **error: [Errno 2] No such file or directory: '/usr/local/cuda-11.7/bin/nvcc'
      [end of output]**
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pytorch3d
  Running setup.py clean for pytorch3d
Failed to build pytorch3d
ERROR: Could not build wheels for pytorch3d, which is required to install pyproject.toml-based projects
```
AND for running the demo setup to **`OVMM`,** I am getting a new error message after installing mlk=2024.0, I think its related to not being able to install `pytorch3d`. 

```
File "/home/che/.conda/envs/home-robot/lib/python3.9/site-packages/**pytorch3d**/ops/ball_query.py", line 10, in <module>
    from pytorch3d import _C
ImportError: libtorch_cuda_cu.so: cannot open shared object file: No such file or directory
```

I currently have Nvidia-smi 550.54.15 with CUDA Version 12.4. I think `pytorch3d` requires cuda-11.7, but this this is running on a container would I still need to downgrade? Any insight on this would be very much appreciated! 
Thank you so much for your help! 

```
(home-robot) che@che-MS-7D99:~/Desktop/home-robot$ nvidia-smi
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.54.15              Driver Version: 550.54.15      CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+

```

yvsriram, 2024-05-03:
Hey, I am not entirely sure about this one but you can try a few things if you haven't already.

For the error you are facing:

> File "/home/che/.conda/envs/home-robot/lib/python3.9/site-packages/**pytorch3d**/ops/ball_query.py", line 10, in <module>
>     from pytorch3d import _C
> ImportError: libtorch_cuda_cu.so: cannot open shared object file: No such file or directory

1. What is the version of your installed pytorch3d? My working environment has `pytorch3d==0.7.5`. You can try pinning to that and see if it helps. 

2. If not done already, it would be good to check that torch and cuda are properly installed and compatible: https://github.com/open-mmlab/mmdetection3d/issues/438#issuecomment-819154481 and also check if any of the solutions from similar issues on [pytorch3d](https://github.com/facebookresearch/pytorch3d/issues/688) are relevant for you? Can you share your `conda env export` so that I can understand your conda environment better?

3. You can try building pytorch3d from source and check if it solves the issue.

> would I still need to downgrade?

We recommend sticking to the CUDA 11.7 and python 3.9. The other dependencies are setup such that they work in this setting. So it may be great if you could downgrade to use cuda 11.7.

Otherwise, it seems like pytorch3d is supported on higher versions of Cuda: https://github.com/facebookresearch/pytorch3d/blob/main/INSTALL.md too? I am not sure if other dependencies and our code are compatible with cuda 12.4.

Do let us know what works and doesn't work for you.

eyucherin, 2024-05-03:
Hello! 
I have checked that my environment has pytorch3d == 0.7.5 

Here is my conda env export. 

```
(home-robot) che@che-MS-7D99:~/home-robot$ cat environment.yml
name: home-robot
channels:
  - pytorch3d
  - pytorch
  - pyg
  - aihabitat
  - nvidia
  - conda-forge
  - defaults
dependencies:
  - _libgcc_mutex=0.1=conda_forge
  - _openmp_mutex=4.5=2_kmp_llvm
  - aiohttp=3.9.5=py39hd1e30aa_0
  - aiosignal=1.3.1=pyhd8ed1ab_0
  - aom=3.9.0=hac33072_0
  - assimp=5.3.1=h8343317_3
  - async-timeout=4.0.3=pyhd8ed1ab_0
  - attrs=23.2.0=pyh71513ae_0
  - blas=2.122=openblas
  - blas-devel=3.9.0=22_linux64_openblas
  - brotli-python=1.1.0=py39h3d6467e_1
  - bzip2=1.0.8=hd590300_5
  - c-ares=1.28.1=hd590300_0
  - ca-certificates=2024.3.11=h06a4308_0
  - cached-property=1.5.2=hd8ed1ab_1
  - cached_property=1.5.2=pyha770c72_1
  - cairo=1.18.0=h3faef2a_0
  - certifi=2024.2.2=pyhd8ed1ab_0
  - cffi=1.16.0=py39h7a31438_0
  - charset-normalizer=3.3.2=pyhd8ed1ab_0
  - cmake=3.29.2=hcfe8598_0
  - colorama=0.4.6=pyhd8ed1ab_0
  - console_bridge=1.0.2=h924138e_1
  - cuda-cudart=11.7.99=0
  - cuda-cupti=11.7.101=0
  - cuda-libraries=11.7.1=0
  - cuda-nvrtc=11.7.99=0
  - cuda-nvtx=11.7.91=0
  - cuda-runtime=11.7.1=0
  - cuda-version=11.7=h67201e3_3
  - cudatoolkit=11.7.1=h4bc3d14_13
  - cudnn=8.9.7.29=hbc23b4c_3
  - cycler=0.12.1=pyhd8ed1ab_0
  - dataclasses=0.8=pyhc8e2a94_3
  - dav1d=1.2.1=hd590300_0
  - eigen=3.4.0=h4bd325d_0
  - exa=0.10.1=habb4d0f_0
  - expat=2.6.2=h59595ed_0
  - ffmpeg=3.2.4=1
  - font-ttf-dejavu-sans-mono=2.37=hab24e00_0
  - font-ttf-inconsolata=3.000=h77eed37_0
  - font-ttf-source-code-pro=2.038=h77eed37_0
  - font-ttf-ubuntu=0.83=h77eed37_1
  - fontconfig=2.14.2=h14ed4e7_0
  - fonts-conda-ecosystem=1=0
  - fonts-conda-forge=1=0
  - freetype=2.12.1=h267a509_2
  - fribidi=1.0.10=h516909a_0
  - frozenlist=1.4.1=py39hd1e30aa_0
  - fsspec=2024.3.1=pyhca7485f_0
  - future=1.0.0=pyhd8ed1ab_0
  - fvcore=0.1.5.post20221221=pyhd8ed1ab_0
  - fzf=0.50.0=ha8f183a_0
[... 376 lines cut ...]
    - tensorboard-data-server==0.6.1
    - tensorboard-plugin-wit==1.8.1
    - tensorboardx==2.6.2.2
    - tifffile==2024.4.24
    - timm==0.9.16
    - tokenizers==0.19.1
    - tomli==2.0.1
    - torchgeometry==0.1.2
    - tqdm==4.65.2
    - traitlets==5.14.3
    - transformers==4.40.1
    - trimesh==3.23.5
    - typer==0.12.3
    - tzdata==2024.1
    - urllib3==1.26.18
    - virtualenv==20.26.1
    - wcwidth==0.2.13
    - webdataset==0.1.40
    - werkzeug==3.0.2
    - widgetsnbextension==4.0.10
    - xtcocotools==1.14.3
    - yapf==0.40.2
    - zipp==3.18.1
prefix: /home/che/.conda/envs/home-robot
```

eyucherin, 2024-05-03:
Whenever I try to download the dataset, my all the content on the browser breaks down and I get this error message. Still not able to download wheels for pytorch3d

[image]

eyucherin, 2024-05-04:
I was able to fix the above by using this command `MAX_JOBS=1 python -m pip install "git+https://github.com/facebookresearch/pytorch3d.git"` when downloading the dataset. Getting help from [this](https://github.com/facebookresearch/pytorch3d/issues/1770#issuecomment-2029785228) issue