facebookresearch/home-robot #493: Sad, I couldn't even add a new rgb sensor to the env...

Yonggie, 2024-05-09:
## 🚀 Feature
Will there be a env yaml tutorial?

## Motivation
Hi, I try to add a new ``rbg sensor`` to the env. I added some codes in exact the same format as the existing:
```
# @package _global_

defaults:
  - /habitat: habitat_config_base
  - /habitat/simulator/agents@habitat.simulator.agents.main_agent: rgbd_agent
  - /habitat/task/ovmm: nav_to_obj
  - /habitat/dataset/ovmm: hssd
  - _self_

habitat:
  gym:
    obs_keys:
      - head_depth
# my new key
      - head_rgb
# my new key ends
      - object_embedding
      - ovmm_nav_goal_segmentation
      - receptacle_segmentation
      - start_receptacle
      - robot_start_gps
      - robot_start_compass
  environment:
    max_episode_steps: 400
  simulator:
    type: OVMMSim-v0
   ...
    agents:
      main_agent:
        radius: 0.3
        height: 1.41
        articulated_agent_urdf: data/robots/hab_stretch/urdf/hab_stretch.urdf
        articulated_agent_type: "StretchRobot"
        ik_arm_urdf: null
        sim_sensors:
# my new config
           head_rgb_sensor:
             height: 640
             width: 480
             hfov: 42
# my new config ends
           head_depth_sensor:
             height: 640
             width: 480
             hfov: 42
             position: [0,1.31,0]
           head_panoptic_sensor:
             height: 640
             width: 480
             hfov: 42
             position: [0,1.31,0]
...
```

When I try to init env with ``with habitat.Env(config=habitat.get_config(cfg_path)) as env:``, it says
```
hydra.errors.ConfigCompositionException: In 'nav_to_obj.yaml': ConfigKeyError raised while composing config:
Key 'hfov' not in 'SimulatorSensorConfig'
    full_key: habitat.simulator.agents.main_agent.sim_sensors.head_rgb_sensor.hfov
    reference_type=SimulatorSensorConfig
    object_type=SimulatorSensorConfig
```

lol, I just wonder how comes it doesn't work... I added a ``rgb sensor``  with the same codes as ``depth sensor``...

Will you kindly help this out with some materials/documents/tutorials/links I can look up for? THX!

Yonggie, 2024-05-09:
Deleted the ``hfov`` and it came with:
```
omegaconf.errors.MissingMandatoryValue: Missing mandatory value: habitat.simulator.agents.main_agent.sim_sensors.head_rgb_sensor.type
    full_key: habitat.simulator.agents.main_agent.sim_sensors.head_rgb_sensor.type
    reference_type=SimulatorSensorConfig
    object_type=SimulatorSensorConfig
```

yvsriram, 2024-05-09:
Just change the default to use:
`/habitat/simulator/agents@habitat.simulator.agents.main_agent: rgbdp_head_rgb_third_agent` instead of `rgbd_agent`. 

You can exclude the `third_rgb_sensor` [here](https://github.com/facebookresearch/habitat-lab/blob/ovmm_main_merge/habitat-lab/habitat/config/habitat/simulator/agents/rgbdp_head_rgb_third_agent.yaml) as it is not needed during training.

Yonggie, 2024-05-09:
Worked, thx~

closed 2024-05-09 by Yonggie