facebookresearch/home-robot #494: Error setting up environment

Welly0902, 2024-05-09:
## 🐛 Bug

I'm trying to try out the GOAT code.
Currently I would like to try out GOAT in simulator without hardware platform (I guess it's the habitat-lab environment)
But failed to do so.

Steps to reproduce the behavior:

When I trying to run the code:
`python goat.py --trajectory=trajectory1 --goals=object_toilet,image_bed1,language_chair4,image_couch1,language_plant1,language_plant2,image_refrigerator1`
I got error like this:
[image]

which "spot_wrapper" doesn't seem like a package that can be installed by pip or conda
I think it's the module in the GOAT code, but I'm not sure what to do to fix this.

(I follow step 5, 6, 7 on [ray_tracing](https://github.com/facebookresearch/home-robot/tree/ray_tracing))
Also in [step 7](https://github.com/facebookresearch/home-robot/blob/ray_tracing/src/home_robot_sim/README.md),
[image]
I got this error:
[image]

## Expected behavior

Not quite sure what exactly would show, but I tried out habitat-sim and habitat-lab and works fine. I would expect there's a Spot robot running in the simulator env and do the GOAT(imagenav) job.

## Additional context

No

yvsriram, 2024-05-09:
Hey if you are using a simulator, this is the right project: https://github.com/facebookresearch/home-robot/tree/goat/projects/habitat_goat. 

But I would recommend to check this repository: https://github.com/Ram81/goat-bench for using the latest dataset and task

Welly0902, 2024-05-19:
> Hey if you are using a simulator, this is the right project: https://github.com/facebookresearch/home-robot/tree/goat/projects/habitat_goat.
> 
> The entry point you are using is for real spot
> 
> But I would recommend to check this repository: https://github.com/Ram81/goat-bench for using the latest dataset and task

Hi thanks for pointing the direction.
I'm trying the above one.
I set up the home robot repo (branch: goat) and follow the above page.
It seems like the previous problems I faced can solved by correctly clone the repo's branch by git command.
After `submodule update` by the step-by-step guide and then can correctly install the package.

However, I have facing new question: 
When I'm trying to run the [above page](https://github.com/facebookresearch/home-robot/tree/goat/projects/habitat_goat) python file `python projects/habitat_goat/eval_episode.py`
I got the error: 
[image]

Where should I place the goat dataset to make the yaml file identify the location?
[image]
I have tried: /path/to/home-robot/data/datasets/goat
and /path/to/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/goat
but both don't work

Thanks!

yvsriram, 2024-05-20:
Hi, can you post the full error message after running with `HYDRA_FULL_ERROR=1`. Did you download the dataset under the path this path: https://github.com/facebookresearch/habitat-lab/blob/018cdd74609efd7f57103a896950c6952b8c09bf/habitat-lab/habitat/config/habitat/dataset/goat/hm3d.yaml#L8C1-L8C82

Welly0902, 2024-05-20:
Hi, I add the file to the location and the previous error was solved.
But it seems like I still have some materials missing.

I add `HYDRA_FULL_ERROR=1` at the end of the command.
And the error like this:

`(bot-goat) blue@blue-System-Product-Name:~/harv/habitat/home-robot$ python projects/habitat_goat/eval_episode.py HYDRA_FULL_ERROR=1
pybullet build time: Nov 28 2023 23:52:03
Traceback (most recent call last):
  File "/home/blue/harv/habitat/home-robot/projects/habitat_goat/eval_episode.py", line 21, in <module>
    from config_utils import get_config
  File "/home/blue/harv/habitat/home-robot/projects/habitat_goat/config_utils.py", line 6, in <module>
    from habitat_baselines.config.default import _BASELINES_CFG_DIR
  File "/home/blue/harv/habitat/home-robot/src/third_party/habitat-lab/habitat-baselines/habitat_baselines/__init__.py", line 14, in <module>
    from habitat_baselines.rl.ppo.ppo_trainer import PPOTrainer, RolloutStorage
  File "/home/blue/harv/habitat/home-robot/src/third_party/habitat-lab/habitat-baselines/habitat_baselines/rl/ppo/ppo_trainer.py", line 33, in <module>
    from habitat_baselines.common.construct_vector_env import construct_envs
  File "/home/blue/harv/habitat/home-robot/src/third_party/habitat-lab/habitat-baselines/habitat_baselines/common/construct_vector_env.py", line 12, in <module>
    from habitat.utils.gym_definitions import make_gym_from_config
  File "/home/blue/harv/habitat/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/utils/gym_definitions.py", line 123, in <module>
    cfg_data = habitat.get_config(full_path)
  File "/home/blue/harv/habitat/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/config/default.py", line 100, in get_config
    cfg = compose(
  File "/home/blue/anaconda3/envs/bot-goat/lib/python3.9/site-packages/hydra/compose.py", line 38, in compose
    cfg = gh.hydra.compose_config(
  File "/home/blue/anaconda3/envs/bot-goat/lib/python3.9/site-packages/hydra/_internal/hydra.py", line 594, in compose_config
    cfg = self.config_loader.load_configuration(
  File "/home/blue/anaconda3/envs/bot-goat/lib/python3.9/site-packages/hydra/_internal/config_loader_impl.py", line 142, in load_configuration
    return self._load_configuration_impl(
  File "/home/blue/anaconda3/envs/bot-goat/lib/python3.9/site-packages/hydra/_internal/config_loader_impl.py", line 253, in _load_configuration_impl
    defaults_list = create_defaults_list(
  File "/home/blue/anaconda3/envs/bot-goat/lib/python3.9/site-packages/hydra/_internal/defaults_list.py", line 745, in create_defaults_list
    defaults, tree = _create_defaults_list(
  File "/home/blue/anaconda3/envs/bot-goat/lib/python3.9/site-packages/hydra/_internal/defaults_list.py", line 715, in _create_defaults_list
    defaults_tree = _create_defaults_tree(
  File "/home/blue/anaconda3/envs/bot-goat/lib/python3.9/site-packages/hydra/_internal/defaults_list.py", line 356, in _create_defaults_tree
    ret = _create_defaults_tree_impl(
  File "/home/blue/anaconda3/envs/bot-goat/lib/python3.9/site-packages/hydra/_internal/defaults_list.py", line 457, in _create_defaults_tree_impl
    return _expand_virtual_root(repo, root, overrides, skip_missing)
  File "/home/blue/anaconda3/envs/bot-goat/lib/python3.9/site-packages/hydra/_internal/defaults_list.py", line 280, in _expand_virtual_root
    subtree = _create_defaults_tree_impl(
  File "/home/blue/anaconda3/envs/bot-goat/lib/python3.9/site-packages/hydra/_internal/defaults_list.py", line 573, in _create_defaults_tree_impl
    add_child(children, new_root)
  File "/home/blue/anaconda3/envs/bot-goat/lib/python3.9/site-packages/hydra/_internal/defaults_list.py", line 520, in add_child
    subtree_ = _create_defaults_tree_impl(
  File "/home/blue/anaconda3/envs/bot-goat/lib/python3.9/site-packages/hydra/_internal/defaults_list.py", line 565, in _create_defaults_tree_impl
    add_child(children, new_root)
  File "/home/blue/anaconda3/envs/bot-goat/lib/python3.9/site-packages/hydra/_internal/defaults_list.py", line 520, in add_child
    subtree_ = _create_defaults_tree_impl(
  File "/home/blue/anaconda3/envs/bot-goat/lib/python3.9/site-packages/hydra/_internal/defaults_list.py", line 488, in _create_defaults_tree_impl
    config_not_found_error(repo=repo, tree=root)
  File "/home/blue/anaconda3/envs/bot-goat/lib/python3.9/site-packages/hydra/_internal/defaults_list.py", line 799, in config_not_found_error
    raise MissingConfigException(
hydra.errors.MissingConfigException: In 'habitat/task/goat': Could not load 'habitat/task/lab_sensors/multigoal_sensor'.

Config search path:
	provider=hydra, path=pkg://hydra.conf
	provider=main, path=file:///home/blue/harv/habitat/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/config/benchmark/nav/goat
	provider=habitat, path=pkg://habitat.config
	provider=schema, path=structured://
`

I think there's still some files i should put in the right place. But I can't find this one 'habitat/task/lab_sensors/multigoal_sensor' anywhere.

yvsriram, 2024-07-12:
Hi, I think you are on the incorrect habitat-lab branch. You can find this sensor defined [here](https://github.com/facebookresearch/habitat-lab/blob/home-robot_goat_support/habitat-lab/habitat/tasks/nav/goat_task.py#L26). I will update the habitat-lab submodule to point to the `home-robot_goat_support` branch