facebookresearch/home-robot #497: Actions of the agent does not take effect to envs on branch ``home-robot-ovmm-challenge-2024``

Yonggie, 2024-05-24:
## 🐛 Bug

On the tag ``home-robot-ovmm-challenge-2024`` branch, the actions of any agent does not take effect to the env no matter what kind of agent I choose.

The visualization stays the same as follows:
[image]

[image]

[image]

My agent keeps saying turn right but the observation from env stays the same:
[image]

Yonggie, 2024-05-27:
I highly suspect that this could be a GPU memory insufficient problem! It gets back to normal the other day! But how come no warnings or exceptions coming out?

GuanxingLu, 2024-05-27:
I have also encountered a similar problem...

yvsriram, 2024-05-27:
The agent wouldn't move if the action takes it outside the navmesh. We can print warnings [here](https://github.com/facebookresearch/habitat-lab/blob/b727ca9f7123101aaedb737ca9ccc1b153525dd9/habitat-lab/habitat/tasks/rearrange/actions/actions.py#L810) if it tries to move outside the navmesh. 

 But @Yonggie it should be able to rotate in the screenshot you shared. I doubt if it is a hardware issue. Can you check if you are using the right habitat-lab [commit](https://github.com/facebookresearch/habitat-lab/tree/b727ca9f7123101aaedb737ca9ccc1b153525dd9) too?

@GuanxingLu were you running the default agent too?

Yonggie, 2024-05-28:
I think it's about the hardware, cus it got back to normal yesterday with the same device and same codes. But I wonder how come there's no warnings or exceptions even when encoutered insufficient hardware requirements?

Fyi, this is the habitat-lab version & commit:
version:
```
habitat-baselines             0.2.5              /share/home-robot/src/third_party/habitat-lab/habitat-baselines
habitat-lab                   0.2.5              /share/home-robot/src/third_party/habitat-lab/habitat-lab
```
git log:
```
commit b727ca9f7123101aaedb737ca9ccc1b153525dd9 (HEAD)
Author: Sriram Yenamandra <yvsriram99@gmail.com>
Date:   Thu Nov 2 12:27:23 2023 -0400

    fix camera_pose update it to be relative to agent start

commit d965bd0264f857845d3d0167806ff17b994c6441
Author: Sriram Yenamandra <yvsriram99@gmail.com>
:...skipping...
commit b727ca9f7123101aaedb737ca9ccc1b153525dd9 (HEAD)
Author: Sriram Yenamandra <yvsriram99@gmail.com>
Date:   Thu Nov 2 12:27:23 2023 -0400

    fix camera_pose update it to be relative to agent start

commit d965bd0264f857845d3d0167806ff17b994c6441
Author: Sriram Yenamandra <yvsriram99@gmail.com>
Date:   Sun Sep 24 15:03:24 2023 -0400

    remove debug statements

commit 10a6e87380ef6eb36e6b2f04af62b74df69a2a7e
Author: Sriram Yenamandra <yvsriram99@gmail.com>
Date:   Sat Sep 23 00:59:21 2023 -0400

    fix: saved ckpts no longer have actor_critic prefix

commit 6af72390cdcae262d63ab8755209b8ed8d2d9272
Merge: aef3de86 17ec6b3c
Author: Sriram Yenamandra <yvsriram99@gmail.com>
Date:   Fri Sep 22 16:09:16 2023 -0400
```