facebookresearch/home-robot #500: Add ``goal_name``/``object_category`` in training env observation?

Yonggie, 2024-05-31:
## 🚀 Feature
To obtain ``goal_name``/ target item name (the item we pick up) from observation.
The ovmm instruction goal is ``"Move " + obj_name + " from " + start_receptacle + " to " + goal_receptacle``, which is only provided by env in eval mode at ``observation.task_observation['object_goal']``.

The Observations from training env is 
```
{
'head_rgb': array(), dtype=uint8), 
'head_depth': array(), dtype=float32), 
'head_panoptic': array(), dtype=int32), 'third_rgb': array(), dtype=uint8), 
'joint': array() dtype=float32), 
'joint_vel': array([0., 0., 0., 0., 0., 0., 0., 0., 0., 0.], dtype=float32), 
'start_receptacle': array([7]), 
'ovmm_nav_goal_segmentation': array(),
'receptacle_segmentation': array(), 
'object_embedding': array(), 
'robot_start_gps': array([-0.,  0.], dtype=float32),
'robot_start_compass': array([1.110223e-16], dtype=float32)
}
```
which only contains receptables info and does not contain target item id. Thus I could only get ``move ? from rec1 to rec2``, which is an incomplete instructions.

I wonder if I can get the target item from observation in training env observation?

## Motivation
To take instruction and image together as input when training new model.

Yonggie, 2024-06-03:
I think I finally find where the ``object_id`` is! Such a hard experience debugging&anlyzing a huge python project!

It's in ``src/third_party/habitat-lab/habitat-lab/habitat/core/embodied_task.py``, the sim env has task and `OVMMEpisode class`` at ``src/third_party/habitat-lab/habitat-lab/habitat/datasets/ovmm/ovmm_dataset.py`` has the ids of object and receptacles. And add new keys of the return observation at ``reset`` function at ``Embodied.reset()``.

Having hard time finding the ``parent/child class`` cus the vscode debugging only leads me to the abstract class instead of the leaf child class. Have to mannually look through all the files with eye balls and mannually search the text to check if the file is related...

yvsriram, 2024-06-03:
Hi, sorry we missed this! 

You can add the `object_category_sensor` to your training config like here:

https://github.com/facebookresearch/habitat-lab/blob/45aa489a84a853cb10b9e7c87383262831a6bd22/habitat-lab/habitat/config/habitat/task/ovmm/ovmm.yaml#L43
 
 https://github.com/facebookresearch/habitat-lab/blob/45aa489a84a853cb10b9e7c87383262831a6bd22/habitat-lab/habitat/config/benchmark/ovmm/ovmm.yaml#L25
 
 https://github.com/facebookresearch/habitat-lab/blob/45aa489a84a853cb10b9e7c87383262831a6bd22/habitat-lab/habitat/tasks/ovmm/ovmm_sensors.py#L75
 
 This will provide the object return the object_id. You can create another sensor or modify it to just return [object_category](https://github.com/facebookresearch/habitat-lab/blob/45aa489a84a853cb10b9e7c87383262831a6bd22/habitat-lab/habitat/tasks/ovmm/ovmm_sensors.py#L75) name

closed 2024-06-04 by Yonggie