facebookresearch/home-robot #501: SparseVoxelMap _show_pytorch3d showing duplicate items from scene

yrraadi-io, 2024-06-01:
I was playing around with creating my own custom `SparseVoxelMapAgent` and using the underlying `SparseVoxelMap` class. At the end of my own custom short episode (hssd scene 790) where i just take a few steps toward the object, I am using the `_show_pytorch3d` method to produce my voxel map 3d visualization and this is what it produces.
[image]
As you can see items are being duplicated and I'm unsure what the source of this issue is. Am I doing something wrong or is this a bug in the implementation in the repository?

yvsriram, 2024-06-02:
Hi @yrraadi-io Awesome progress! I think we had to use this [transform](https://github.com/facebookresearch/home-robot/blame/ede6a67a2d0c0c8e12ad3b9726f330cea6cf90f5/src/home_robot_sim/home_robot_sim/env/habitat_ovmm_env/habitat_ovmm_env.py#L152) to get the pointclouds oriented:

yrraadi-io, 2024-06-02:
> transform

Hi, thanks for sharing. Just to clarify I believe that transform is already being applied since when we `self._env.reset` the environment or  call `self._env.apply_action` internally `self._preprocess_obs(habitat_obs)` is called which applies the tranformation to the `camera_pose`.

Are you suggesting we need to re-apply the same tranformation before calling the `step` method in my custom `SparseVoxelMapAgent`?

yvsriram, 2024-06-02:
It uses `convert_pose_to_real_world_axis`. I was suggesting to try using `use_opencv_camera_pose`

yrraadi-io, 2024-06-02:
that worked, thank you! Just out of curiosity why does this work and not the other tranformation?

[image]

yvsriram, 2024-06-02:
Amazing! Just that sparse voxel code expects opencv's camera convention, different from what habitat uses.