facebookresearch/home-robot #503: "None of the goals are reachable from current position for episode {episode_id}"

Yonggie, 2024-06-04:
## 🚀 Feature
Hi, in training phase I had this problem. In what scenario will this come up?
A wrong ``goal id`` that doesn't exist in the current episode? Or the goal is misplaced and the agent can't find a valid path?

It's at ``src/third_party/habitat-lab/habitat-lab/habitat/tasks/ovmm/utils.py`` line ``26``.

yvsriram, 2024-06-04:
This error is not expected. This could be a navmesh issue. Can you check if you are checked out to the `ovmm` branch at `data/hssd-hab`? Otherwise, your code will generate navmeshes under `data/hssd-hab/navmeshes/scenes-uncluttered/stretch_robot`, which may not be compatible with the agent initialisations. It will only work with [precomputed navmeshes](https://huggingface.co/datasets/hssd/hssd-hab/tree/ovmm/navmeshes/scenes-uncluttered/stretch_robot).

Yonggie, 2024-06-05:
- ``git branch`` I got ``main*``, 
- ``git status`` got tons of deleted items in green (too much charactars), paste some as example:
```
On branch main
Your branch is up to date with 'origin/main'.

Changes to be committed:
  (use "git restore --staged <file>..." to unstage)
	deleted:    .gitattributes
	deleted:    .gitignore
	deleted:    README.md
	deleted:    hssd-hab-uncluttered.scene_dataset_config.json
	deleted:    hssd-hab.scene_dataset_config.json
	deleted:    objects/0/0001fb06b075a743e6289236cf049df3ad5dfa9c.collider.glb
	deleted:    objects/0/0001fb06b075a743e6289236cf049df3ad5dfa9c.glb
	deleted:    objects/0/0001fb06b075a743e6289236cf049df3ad5dfa9c.object_config.json
	deleted:    objects/0/000bbe302b53fd2904e7ae92e1516a18f29de02d.collider.glb
	deleted:    objects/0/000bbe302b53fd2904e7ae92e1516a18f29de02d.glb
        ...
```
- ``git log`` I got 
```
commit 9ec06b0d25cba85b6ee422b02e52e14ec4acafc7 (HEAD -> main, tag: v0.2.5, origin/main, origin/HEAD)
Author: Alexander Clegg <aclegg3@users.noreply.huggingface.co>
Date:   Fri Nov 10 18:17:32 2023 +0000

    v0.2.5 README update

commit fa25a62b83b2644fbff16ea55d0125c3b389fb58
Author: aclegg3 <alexanderwclegg@gmail.com>
Date:   Fri Nov 10 10:15:28 2023 -0800

    v0.2.5 update 2: updated object configs, filter files, uncluttered scenes
```

closed 2024-06-07 by Yonggie