facebookresearch/home-robot #509: Home Robot Spot Support

trangml-mit, 2024-07-01:
## 🐛 Bug
I was attempting to follow the [instructions to setup spot](https://github.com/facebookresearch/home-robot/tree/main/src/home_robot_spot) in home-robot, but ran into the issue that the [fork](https://github.com/jdvakil/spot-sim2real) that is described in the readme is private.

Additionally, when following the setup instructions, the default commit for the facebookresearch version of spot-sim2real also does not seem to exist.

## Steps to Reproduce

Steps to reproduce the behavior:

```
git clone https://github.com/facebookresearch/home-robot.git --recursive
```
output

> Submodule path 'src/third_party/habitat-lab': checked out 'b727ca9f7123101aaedb737ca9ccc1b153525dd9'
fatal: remote error: upload-pack: not our ref 1cecb34f5a3e52119baec35bbc9d48bf48e06431
Fetched in submodule path 'src/third_party/spot-sim2real', but it did not contain 1cecb34f5a3e52119baec35bbc9d48bf48e06431. Direct fetching of that commit failed.

## Expected behavior

No errors when cloning, or a reference to a public fork.

## Additional context

I'd like to run the habitat-ovmm challenge with the spot as the robot platform.

trangml-mit, 2024-07-01:
I see this issue [here](https://github.com/facebookresearch/home-robot/pull/495), which seems to fix the submodule error, but the commit error still exists. Also, if the guidance is to work of the main branch of facebookresearch/spot-sim2real, can the readme be updated?

Jdvakil, 2024-07-01:
Hi, this [doc](https://github.com/facebookresearch/home-robot/blob/main/docs/spot.md) should have the latest instructions.   The private commit bug should be fixed

trangml-mit, 2024-07-03:
Thanks for the doc! I was able to install the rest of the package. I am running into the issue now of actually trying to run some of the example code. Is there example code that runs the spot in simulation as a replacement to the Stretch in the OVMM task, or is all the example code intended to be run on the real robot?

I'm not able to run `src/home_robot_spot/examples/demo.py` without the following error:

```
$ python src/home_robot_spot/examples/demo.py 
Make sure path to accel-cortex base folder is set in the ACCEL_CORTEX environment variable.
If you do not know what that means, this code is not for you!
Traceback (most recent call last):
  File "/data2/home-robot/src/home_robot_spot/examples/demo.py", line 24, in <module>
    from examples.demo_utils.mock_agent import MockSpotDemoAgent
  File "/data2/home-robot/src/home_robot_spot/examples/demo_utils/mock_agent.py", line 46, in <module>
    from home_robot.utils.rpc import (
  File "/data2/home-robot/src/home_robot/home_robot/utils/rpc.py", line 26, in <module>
    raise (e)
  File "/data2/home-robot/src/home_robot/home_robot/utils/rpc.py", line 14, in <module>
    sys.path.append(os.path.expanduser(os.environ["ACCEL_CORTEX"]))
  File "/home/ma32004/mambaforge/envs/hr/lib/python3.9/os.py", line 679, in __getitem__
    raise KeyError(key) from None
KeyError: 'ACCEL_CORTEX'
```

Jdvakil, 2024-07-05:
Oh, I know this error, I will test it out on my spot and get back to you.

joanne40226, 2025-06-06:
@trangml-mit 
hi, i ran into the same problem, i wonder if you solved it already?
thanks.