facebookresearch/home-robot #516: error in viz_obj.translation = mn.Vector3(*position)

Beat-in-our-hearts, 2024-09-09:
1. When I run the OVMM 'evaluate' script, 

```
#/bin/bash

export MAGNUM_LOG=quiet
export HABITAT_SIM_LOG=quiet

set -x
python -u -m habitat_baselines.run --config-name=ovmm/rl_discrete_skill_nav_obj.yaml habitat_baselines.evaluate=True
```

I got the following error:
```
Traceback (most recent call last):
  File "/home/lzx/miniforge3/envs/home-robot/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/home/lzx/miniforge3/envs/home-robot/lib/python3.9/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/lzx/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/vector_env.py", line 268, in _worker_env
    observations = env.reset()
  File "/home/lzx/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/gym/gym_env_episode_count_wrapper.py", line 61, in reset
    return self.env.reset(**kwargs)
  File "/home/lzx/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/gym/gym_env_obs_dict_wrapper.py", line 36, in reset
    return self.env.reset(**kwargs)
  File "/home/lzx/miniforge3/envs/home-robot/lib/python3.9/site-packages/gym/core.py", line 283, in reset
    return self.env.reset(**kwargs)
  File "/home/lzx/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/gym/gym_wrapper.py", line 312, in reset
    obs = self.env.reset(*args, return_info=return_info, **kwargs)
  File "/home/lzx/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/environments.py", line 62, in reset
    return super().reset(*args, return_info=return_info, **kwargs)
  File "/home/lzx/miniforge3/envs/home-robot/lib/python3.9/contextlib.py", line 79, in inner
    return func(*args, **kwds)
  File "/home/lzx/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/env.py", line 424, in reset
    observations = self._env.reset()
  File "/home/lzx/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/core/env.py", line 262, in reset
    observations = self.task.reset(episode=self.current_episode)
  File "/home/lzx/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/tasks/ovmm/sub_tasks/nav_to_obj_task.py", line 53, in reset
    obs = super().reset(episode)
  File "/home/lzx/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/tasks/rearrange/sub_tasks/nav_to_obj_task.py", line 288, in reset
    sim.viz_ids["nav_targ_pos"] = sim.visualize_position(
  File "/home/lzx/home-robot/src/third_party/habitat-lab/habitat-lab/habitat/tasks/rearrange/rearrange_sim.py", line 995, in visualize_position
    viz_obj.translation = mn.Vector3(*position)
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. _magnum.Vector3(arg0: _magnum.Vector3ui)
    2. _magnum.Vector3(arg0: _magnum.Vector3i)
    3. _magnum.Vector3(arg0: _magnum.Vector3d)
    4. _magnum.Vector3(arg0: buffer)
    5. _magnum.Vector3()
    6. _magnum.Vector3(arg0: float)
    7. _magnum.Vector3(arg0: float, arg1: float, arg2: float)
    8. _magnum.Vector3(arg0: _magnum.Vector2, arg1: float)
    9. _magnum.Vector3(arg0: Tuple[float, float, float])
```

2. I tried to print postion.shape, found that each environment generates a different size, for example:

```
=========================
pisition.shape:(6, 3)
=========================
=========================
pisition.shape:(16, 3)
=========================
=========================
pisition.shape:(83, 3)
=========================
=========================
pisition.shape:(27, 3)
=========================
```
3. I wrote a simple program code using jupyter in same conda environment:
```
import magnum as mn
import numpy as np

data = np.random.rand(10,3)
mn.Vector3(*data)
```
The same error occurred here
```
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. _magnum.Vector3(arg0: _magnum.Vector3ui)
    2. _magnum.Vector3(arg0: _magnum.Vector3i)
    3. _magnum.Vector3(arg0: _magnum.Vector3d)
    4. _magnum.Vector3(arg0: buffer)
    5. _magnum.Vector3()
    6. _magnum.Vector3(arg0: float)
    7. _magnum.Vector3(arg0: float, arg1: float, arg2: float)
    8. _magnum.Vector3(arg0: _magnum.Vector2, arg1: float)
    9. _magnum.Vector3(arg0: Tuple[float, float, float])

Invoked with: array([0.24895031, 0.86328894, 0.96980985]), array([0.59404719, 0.48294977, 0.37594189]), ................
```

4. I looked at the variables passed to the upper function. in class DynNavRLEnv(RearrangeTask), but still not solve the problem.
```
sim.viz_ids["nav_targ_pos"] = sim.visualize_position(
self._nav_to_info.nav_goal_pos,
sim.viz_ids["nav_targ_pos"],
R = 0.2,
)
```

_**NOTE:**_ Another bug is that even though I changed ‘num_environments: 1’  in the yaml file ‘ovmm/rl_discrete_skill_nav_obj.yaml’, it still generated 12 environments (the same as in the training).
```
import torch
ckpt = torch.load('data/new_checkpoints/nav_to_obj/ckpt.0.pth')
print(ckpt.keys())
print(ckpt['config']['habitat_baselines']['num_environments'])
```
output: 
```
dict_keys(['state_dict', 'config', 'extra_state'])
12
```

I think the evaluate script directly calls config in.pth, and I think it should allow different number of environments for training and validation

Beat-in-our-hearts, 2024-09-09:
I solved the problem by commenting this line of code, but I'm not sure if this line of code matters.

yvsriram, 2024-10-02:
For the first error, yes it is ok to comment this line out. We did not use this visualization.

> I think the evaluate script directly calls config in.pth, and I think it should allow different number of environments for training and validation

Yes, it seems like it is trying to use the config from checkpoint. Can you try `habitat_baselines.load_resume_state_config = False`?