facebookresearch/home-robot #518: There is a problem in the Habitat ObjectNav

jingchuanDDD, 2024-10-14:
## 🐛 Bug

When I run python projects/habitat_objectnav/eval_episode.py,an error occurred：
[image]

## Steps to Reproduce

Steps to reproduce the behavior:
just set up the home-robot,run comands in Habitat ObjectNav and run evaluation.
[image]

## Expected behavior

Is it the problem of the dataset?The setup commands for hm3d dataset are not clear.

## Additional context

I checked this variable and found that it was not updated and was always none

JiahuaPei, 2025-04-21:
the same problem
"""    
def _preprocess_obs(
        self, habitat_obs: habitat.core.simulator.Observations
    ) -> home_robot.core.interfaces.Observations:
        depth = self._preprocess_depth(habitat_obs["depth"])
        goal_id, goal_name = self._preprocess_goal(habitat_obs["objectgoal"])
        obs = home_robot.core.interfaces.Observations(
            rgb=habitat_obs["rgb"],
            depth=depth,
            compass=habitat_obs["compass"],
            gps=self._preprocess_xy(habitat_obs["gps"]),
            task_observations={
                "object_goal": goal_id,
                "goal_name": goal_name,
                "recep_goal": None,
            },
            camera_pose=None,
            third_person_image=None,
        )
        obs = self._preprocess_semantic_and_instance(obs, habitat_obs["semantic"])
        return obs
"""

JiahuaPei, 2025-04-21:
I have solved that.
just like the comment above,camera_pose is set to None forever.
what we should do is to compute the camera_pose by ourselves.
try this code will solve the problem.(but you will still encounter the problem of visualization,just comment it)
home-robot/src/home_robot_sim/home_robot_sim/env/habitat_objectnav_env/habitat_objectnav_env.py
```
import numpy as np
import torch
from scipy.spatial.transform import Rotation as ScipyRotation

def calculate_camera_pose(gps: np.ndarray, compass: np.ndarray, camera_height: float) -> torch.Tensor:
    """
    Calculate an approximate camera_pose (4x4 matrix) based on GPS, Compass, and camera height.
    Use scipy.spatial.transform.Rotation to build the rotation matrix.
    Assumptions: Flat ground, camera has no roll/pitch, Z-axis is up, Compass provides Yaw (radians) around the Z-axis.

    Args:
        gps: Robot's XY position in the world coordinate system (shape: (2,)).
        compass: Robot's yaw angle in the world coordinate system (radians) (shape: (1,)).
                 Assumes 0 radians points towards the positive world X-axis, positive angle means counter-clockwise rotation around the Z-axis.
        camera_height: Fixed camera height (meters).

    Returns:
        A (4, 4) torch.Tensor representing the camera pose in the world coordinate system.
    """
    # 1. Extract position (translation vector T)
    world_x = gps[0]
    world_y = gps[1]
    world_z = camera_height
    translation_vector = np.array([world_x, world_y, world_z], dtype=np.float32)

    # 2. Extract orientation (rotation matrix R)
    yaw = compass[0]
    # Assume Roll and Pitch are 0
    roll = 0.0
    pitch = 0.0

    try:
        # Use 'zyx' order: Yaw (Z), Pitch (Y), Roll (X) - this is the intrinsic rotation order
        # For pure Yaw rotation, 'ZYX' (extrinsic) gives the same result
        # degrees=False indicates input is in radians
        rotation = ScipyRotation.from_euler('zyx', [yaw, pitch, roll], degrees=False)
        rotation_matrix = rotation.as_matrix().astype(np.float32) # Get the 3x3 rotation matrix R
    except Exception as e:
        print(f"Warning: Could not create rotation matrix from Euler angles: {e}. Using identity matrix.")
        # If an error occurs (e.g., Scipy not installed or invalid input), fall back to the identity matrix
        rotation_matrix = np.identity(3, dtype=np.float32)

    # 3. Combine into a 4x4 homogeneous transformation matrix
    camera_pose_np = np.identity(4, dtype=np.float32)
    camera_pose_np[:3, :3] = rotation_matrix
    camera_pose_np[:3, 3] = translation_vector

    # 4. Convert to PyTorch Tensor
    camera_pose_tensor = torch.from_numpy(camera_pose_np)

    return camera_pose_tensor

```
and replace the function
```
    def _preprocess_obs(
        self, habitat_obs: habitat.core.simulator.Observations
    ) -> home_robot.core.interfaces.Observations:
        depth = self._preprocess_depth(habitat_obs["depth"])
        goal_id, goal_name = self._preprocess_goal(habitat_obs["objectgoal"])
        camera_pose = calculate_camera_pose(
            self._preprocess_xy(habitat_obs["gps"]),
            habitat_obs["compass"],
            0.88 
        )
        obs = home_robot.core.interfaces.Observations(
            rgb=habitat_obs["rgb"],
            depth=depth,
            compass=habitat_obs["compass"],
            gps=self._preprocess_xy(habitat_obs["gps"]),
            task_observations={
                "object_goal": goal_id,
                "goal_name": goal_name,
                "recep_goal": None,
            },
            camera_pose=camera_pose,
            third_person_image=None,
        )
        obs = self._preprocess_semantic_and_instance(obs, habitat_obs["semantic"])
        return obs
```