facebookresearch/home-robot #522: Questions about evaluating configuration

Abc12128, 2024-11-17:
Hi, how are you? 
I'd like to know how None, shown in Table 3 of the paper, is implemented in code (with RL and Heuristic and None below the skills in the table). Did you comment out the content below the corresponding skill or modify the type parameter or implement it in some other way?
I have provided the details of Table 3 and the code for the GAZE section below for your convenience.
[image]
[image]

yvsriram, 2024-11-17:
Hello,

You can skip skills like this: https://github.com/facebookresearch/home-robot/blob/ede6a67a2d0c0c8e12ad3b9726f330cea6cf90f5/projects/habitat_ovmm/configs/agent/heuristic_agent.yaml#L163 

We specify a set of skills to skip in the config.

Abc12128, 2024-11-18:
> Hello,
> 
> You can skip skills like this:
> 
> https://github.com/facebookresearch/home-robot/blob/ede6a67a2d0c0c8e12ad3b9726f330cea6cf90f5/projects/habitat_ovmm/configs/agent/heuristic_agent.yaml#L163
> 
> We specify a set of skills to skip in the config.

OK thank you for your reply

closed 2024-11-18 by Abc12128