facebookresearch/home-robot #527: None of the GOAT branches work

aminkashiri, 2025-03-12:
I am trying to run the code for the paper "GOAT: GO to Any Thing" in simulations. I've read all related Github issues, but I am still facing some problems. My objective is to run `projects/habitat_goat/eval_episode.py`, which I assume is this paper's implementation in habitat. I have the following problems, and please correct me if I'm wrong:

1. Which of the branches should I try? 
"goat-sim" branch seems broken. The related issue is [here](https://github.com/facebookresearch/home-robot/issues/450), which says a file is missing ("/srv/flash1/rramrakhya3/fall_2023/goat/data/hm3d_meta/ovon_categories_final_split.json"). The final comment refers to the [goat-bench](https://github.com/Ram81/goat-bench) repo. 
However, goat-bench repo only includes the detail on how to run the baselines SenseAct-NN Monolithic Baseline and SenseAct-NN Skill-Chain Baseline, and not the original GOAT (known as modular GOAT). In one of the [issues](https://github.com/Ram81/goat-bench/issues/6), they have said: "the codebase for modular GOAT baseline is part of the home robot codebase in [this branch](https://github.com/facebookresearch/home-robot/commits/goat-sim/). However this branch is not actively maintained. We are planning to clean the codebase and release but that might take some more time.", and they refer back to "goat-sim" branch (circular refers).
Another branch is the "goat" branch itself, which is not working either. If this is the correct branch that I should use, I can create another issue and explain the problem there. 

2. What datasets should I use? 
For the scene dataset, we need the full HM3D dataset, for which we should use the instructions [here](https://github.com/facebookresearch/habitat-sim/blob/main/DATASETS.md#habitat-matterport-3d-research-dataset-hm3d). 
For the task dataset (is it the same with "episode dataset"?), I am finding different sources. The [readme](https://github.com/facebookresearch/home-robot/tree/goat-sim/projects/habitat_goat)  in habitat_goat project on habitat-sim branch provides a link to [this](https://www.dropbox.com/scl/fi/z0ogqj5cp03qu888w37sw/goat_dataset_v0.zip?rlkey=t0biq6sc88rrxyisuxcfvm3s8&e=1) dataset, which has folders "v0.2" and "v0.3", and inside them is only a train dataset. In habitat-lab's "home-robot_goat_support" branch, the file `habitat-lab/habitat-lab/habitat/config/habitat/dataset/goat/hm3d.yaml` needs the file `data/datasets/goat_openvocab/hm3d/v0.1.2_fixed/{split}/{split}.json.gz`, which I am not sure if it is the same as the dataset I mentioned. Should I just rename files and folders?
To make things more confusing, this [issue](https://github.com/facebookresearch/home-robot/issues/505#issuecomment-2237146892), the talks about `data/datasets/goat/hm3d/v0.3/{split}/{split}.json.gz` but I don't find any config file that requires this in habitat-lab.
There is also [this](https://drive.google.com/file/d/1N0UbpXK3v7oTphC4LoDqlNeMHbrwkbPe/view) dataset provided in the [goat-bench](https://github.com/Ram81/goat-bench?tab=readme-ov-file) repo, which is different from the mentioned datasets.

cybertianli, 2025-05-29:
您好，请问您是否找到模块化 GOAT并复现其代码？

aminkashiri, 2025-06-10:
For anyone trying to run the goat code, this is what I do:
1. I am using the goat-sim branch, and to fix the error of "ovon_categories_final_split.json" missing see [here](https://github.com/facebookresearch/home-robot/issues/534#issuecomment-2959615635).
2. For the dataset, instead I am using [this](https://drive.google.com/file/d/1N0UbpXK3v7oTphC4LoDqlNeMHbrwkbPe/view) dataset provided in the [goat-bench](https://github.com/Ram81/goat-bench?tab=readme-ov-file) repo.

Let me know if you have any issues running the goat in simulations.