facebookresearch/home-robot #530: Unable to submit docker image to EvalAI

namwoam, 2025-03-30:
## 🐛 Bug

After building a baseline solution and successfully testing it using ./scripts/test_local.sh (which produced the expected results), I am unable to submit my Docker image to EvalAI. I have already registered for the challenge and am using the provided authentication token on my local machine.

## Steps to Reproduce

Steps to reproduce the behavior:

1. Follow the instructions provided in the official documentation[https://github.com/facebookresearch/home-robot/blob/home-robot-ovmm-challenge-2024/docs/challenge.md].
2. Attempt to submit the Docker image to EvalAI.
The following error occurs during submission, indicating an issue with AWS credentials:

```
500 Server Error: Internal Server Error for url: https://eval.ai/api/challenges/phases/4522/participant_team/aws/credentials/
```

## Expected behavior

The Docker image should be submitted successfully to the EvalAI leaderboard.

## Additional context

The local machine runs on evalai client python sdk version 1.3.18 and python version 3.10.12

[image]

yvsriram, 2025-04-27:
Hello, unfortunately, the EvalAI leaderboard is no longer supported.

namwoam, 2025-04-28:
Thanks with the update!

closed 2025-04-28 by namwoam