facebookresearch/home-robot #540: An error "Unexpected keys in state_dict" occurred.

WhyxyxIsNotAnAvailableName, 2025-07-07:
## 🐛 Bug

## Steps to Reproduce
I followed the instructions in "[home-robot/src/home_robot_sim/README.md](https://github.com/facebookresearch/home-robot/blob/main/src/home_robot_sim/README.md#environment-setup)" to set up the sim environment. And for the skill _"gaze"_, I would like to use the RL method. So I changed the values of _"skip_skills.gaze_at_obj"_ and _"skip_skills.gaze_at_rec"_ in the file _"home-robot/projects/habitat_ovmm/configs/agent/heuristic_agent.yaml"_ to False. And I run the command:

```
python projects/habitat_ovmm/eval_baselines_agent.py
```

At this point, the following error occurred:

```
Traceback (most recent call last):
  File "/root/autodl-tmp/home-robot/projects/habitat_ovmm/eval_baselines_agent.py", line 117, in <module>
    agent = OpenVocabManipAgent(agent_config, device_id=device_id)
  File "/root/autodl-tmp/home-robot/src/home_robot/home_robot/agent/ovmm_agent/ovmm_agent.py", line 105, in __init__
    self.gaze_agent = PPOAgent(
  File "/root/autodl-tmp/home-robot/src/home_robot/home_robot/agent/ovmm_agent/ppo_agent.py", line 137, in __init__
    {k: self.obs_space[0].spaces[k] for k in self.skill_obs_keys}
  File "/root/autodl-tmp/home-robot/src/home_robot/home_robot/agent/ovmm_agent/ppo_agent.py", line 137, in <dictcomp>
    {k: self.obs_space[0].spaces[k] for k in self.skill_obs_keys}
KeyError: 'robot_head_depth'
```

So I changed the parameter _"gym_obs_keys: robot_head_depth"_ in the file _"home-robot/projects/habitat_ovmm/configs/agent/heuristic_agent.yaml"_ to _"gym_obs_keys: head_depth"_, and then the following error occurred:

```
Traceback (most recent call last):
  File "/root/autodl-tmp/home-robot/projects/habitat_ovmm/eval_baselines_agent.py", line 117, in <module>
    agent = OpenVocabManipAgent(agent_config, device_id=device_id)
  File "/root/autodl-tmp/home-robot/src/home_robot/home_robot/agent/ovmm_agent/ovmm_agent.py", line 105, in __init__
    self.gaze_agent = PPOAgent(
  File "/root/autodl-tmp/home-robot/src/home_robot/home_robot/agent/ovmm_agent/ppo_agent.py", line 193, in __init__
    self.actor_critic.load_state_dict(ckpt["state_dict"])
  File "/root/autodl-tmp/miniconda3/envs/home-robot/lib/python3.9/site-packages/torch/nn/modules/module.py", line 1671, in load_state_dict
    raise RuntimeError('Error(s) in loading state_dict for {}:\n\t{}'.format(
RuntimeError: Error(s) in loading state_dict for PointNavResNetPolicy:
        Unexpected key(s) in state_dict: "net.start_receptacles_embedding.weight". 
        size mismatch for net.visual_encoder.backbone.conv1.0.weight: copying a param with shape torch.Size([32, 3, 7, 7]) from checkpoint, the shape in current model is torch.Size([32, 2, 7, 7]).
        size mismatch for net.state_encoder.rnn.weight_ih_l0: copying a param with shape torch.Size([2048, 1098]) from checkpoint, the shape in current model is torch.Size([2048, 1070]).
```

May this be related to my _"src/third_party/habitat-lab"_ version? I can provide the following information:

```
(home-robot) root@autodl-container-10724d95ce-21b86fe2:~/autodl-tmp/home-robot/src/third_party/habitat-lab# git status
HEAD detached at b727ca9f7
```

## Expected behavior

When my _"gaze"_ skill uses the _"heuristic"_ method or when _"skip_skills.gaze_at_obj"_ is set to True, the evaluation can be carried out normally, and the _"metric"_ and various indicators (success rate) will be output.
I would like the expected behavior to be that the _"gaze"_ skill also performs the evaluation normally when using the _"RL"_ method.

yvsriram, 2025-12-03:
Hi, please try this config instead: https://github.com/facebookresearch/home-robot/blob/main/projects/habitat_ovmm/configs/agent/rl_agent.yaml.

sxyxs, 2025-12-03:
Hi @yvsriram , thanks for the quick reply. My issue is mainly with the checkpoint's shape. I am using the official Docker with  the ckeckpoint from #465. After I use the new config, I **still get the same error message** ```RuntimeError: Error(s) in loading state_dict for PointNavResNetPolicy:
        Missing key(s) in state_dict: "net.visual_encoder.running_mean_and_var._mean", "net.visual_encoder.running_mean_and_var._var", "net.visual_encoder.running_mean_and_var._count". 
        Unexpected key(s) in state_dict: "net.start_receptacles_embedding.weight". 
        size mismatch for net.visual_encoder.backbone.conv1.0.weight: copying a param with shape torch.Size([32, 3, 7, 7]) from checkpoint, the shape in current model is torch.Size([32, 2, 7, 7]).
        size mismatch for net.state_encoder.rnn.weight_ih_l0: copying a param with shape torch.Size([2048, 1098]) from checkpoint, the shape in current model is torch.Size([2048, 1070]).```